OpenCyphal-Garage/platform_specific_components #2: Build issue with STM32L4 series

JeremySavonet, 2019-03-21:
Hi,

It seems that STM32L4 series have no APB1ENR register causing the build to fail.
Here is a dump of the issue:

Compiling uc_stm32_thread.cpp
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp: In function 'void uavcan_stm32::clock::init()':
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:41:39: error: 'struct RCC_TypeDef' has no member named 'APB1ENR'; did you mean 'AHB1ENR'?
 #  define TIMX_RCC_ENR           RCC->APB1ENR
                                       ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:129:5: note: in expansion of macro 'TIMX_RCC_ENR'
     TIMX_RCC_ENR |= TIMX_RCC_ENR_MASK;
     ^~~~~~~~~~~~
In file included from 3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:7:0:
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:43:53: error: 'RCC_APB1ENR_TIM7EN' was not declared in this scope
 #  define TIMX_RCC_ENR_MASK      UAVCAN_STM32_GLUE3(RCC_APB1ENR_TIM,  UAVCAN_STM32_TIMER_NUMBER, EN)
                                                     ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/internal.hpp:92:41: note: in definition of macro 'UAVCAN_STM32_GLUE3_'
 #define UAVCAN_STM32_GLUE3_(A, B, C)    A##B##C
                                         ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:43:34: note: in expansion of macro 'UAVCAN_STM32_GLUE3'
 #  define TIMX_RCC_ENR_MASK      UAVCAN_STM32_GLUE3(RCC_APB1ENR_TIM,  UAVCAN_STM32_TIMER_NUMBER, EN)
                                  ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:129:21: note: in expansion of macro 'TIMX_RCC_ENR_MASK'
     TIMX_RCC_ENR |= TIMX_RCC_ENR_MASK;
                     ^~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:43:53: note: suggested alternative: 'RCC_APB1ENR1_TIM7EN'
 #  define TIMX_RCC_ENR_MASK      UAVCAN_STM32_GLUE3(RCC_APB1ENR_TIM,  UAVCAN_STM32_TIMER_NUMBER, EN)
                                                     ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/internal.hpp:92:41: note: in definition of macro 'UAVCAN_STM32_GLUE3_'
 #define UAVCAN_STM32_GLUE3_(A, B, C)    A##B##C
                                         ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:43:34: note: in expansion of macro 'UAVCAN_STM32_GLUE3'
 #  define TIMX_RCC_ENR_MASK      UAVCAN_STM32_GLUE3(RCC_APB1ENR_TIM,  UAVCAN_STM32_TIMER_NUMBER, EN)
                                  ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:129:21: note: in expansion of macro 'TIMX_RCC_ENR_MASK'
     TIMX_RCC_ENR |= TIMX_RCC_ENR_MASK;
                     ^~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:42:39: error: 'struct RCC_TypeDef' has no member named 'APB1RSTR'; did you mean 'AHB1RSTR'?
 #  define TIMX_RCC_RSTR          RCC->APB1RSTR
                                       ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:130:5: note: in expansion of macro 'TIMX_RCC_RSTR'
     TIMX_RCC_RSTR |=  TIMX_RCC_RSTR_MASK;
     ^~~~~~~~~~~~~
In file included from 3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:7:0:
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:44:53: error: 'RCC_APB1RSTR_TIM7RST' was not declared in this scope
 #  define TIMX_RCC_RSTR_MASK     UAVCAN_STM32_GLUE3(RCC_APB1RSTR_TIM, UAVCAN_STM32_TIMER_NUMBER, RST)
                                                     ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/internal.hpp:92:41: note: in definition of macro 'UAVCAN_STM32_GLUE3_'
 #define UAVCAN_STM32_GLUE3_(A, B, C)    A##B##C
                                         ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:44:34: note: in expansion of macro 'UAVCAN_STM32_GLUE3'
 #  define TIMX_RCC_RSTR_MASK     UAVCAN_STM32_GLUE3(RCC_APB1RSTR_TIM, UAVCAN_STM32_TIMER_NUMBER, RST)
                                  ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:130:23: note: in expansion of macro 'TIMX_RCC_RSTR_MASK'
     TIMX_RCC_RSTR |=  TIMX_RCC_RSTR_MASK;
                       ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:44:53: note: suggested alternative: 'RCC_APB1RSTR1_TIM7RST'
 #  define TIMX_RCC_RSTR_MASK     UAVCAN_STM32_GLUE3(RCC_APB1RSTR_TIM, UAVCAN_STM32_TIMER_NUMBER, RST)
                                                     ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/internal.hpp:92:41: note: in definition of macro 'UAVCAN_STM32_GLUE3_'
 #define UAVCAN_STM32_GLUE3_(A, B, C)    A##B##C
                                         ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:44:34: note: in expansion of macro 'UAVCAN_STM32_GLUE3'
 #  define TIMX_RCC_RSTR_MASK     UAVCAN_STM32_GLUE3(RCC_APB1RSTR_TIM, UAVCAN_STM32_TIMER_NUMBER, RST)
                                  ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:130:23: note: in expansion of macro 'TIMX_RCC_RSTR_MASK'
     TIMX_RCC_RSTR |=  TIMX_RCC_RSTR_MASK;
                       ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:42:39: error: 'struct RCC_TypeDef' has no member named 'APB1RSTR'; did you mean 'AHB1RSTR'?
 #  define TIMX_RCC_RSTR          RCC->APB1RSTR
                                       ^
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp:131:5: note: in expansion of macro 'TIMX_RCC_RSTR'
     TIMX_RCC_RSTR &= ~TIMX_RCC_RSTR_MASK;
     ^~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp: In static member function 'static void uavcan_stm32::CanDriver::initOnce()':
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:996:14: error: 'struct RCC_TypeDef' has no member named 'APB1ENR'; did you mean 'AHB1ENR'?
         RCC->APB1ENR  |=  RCC_APB1ENR_CAN1EN;
              ^~~~~~~
              AHB1ENR
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:996:27: error: 'RCC_APB1ENR_CAN1EN' was not declared in this scope
         RCC->APB1ENR  |=  RCC_APB1ENR_CAN1EN;
                           ^~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:996:27: note: suggested alternative: 'RCC_APB1ENR1_CAN1EN'
         RCC->APB1ENR  |=  RCC_APB1ENR_CAN1EN;
                           ^~~~~~~~~~~~~~~~~~
                           RCC_APB1ENR1_CAN1EN
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:997:14: error: 'struct RCC_TypeDef' has no member named 'APB1RSTR'; did you mean 'AHB1RSTR'?
         RCC->APB1RSTR |=  RCC_APB1RSTR_CAN1RST;
              ^~~~~~~~
              AHB1RSTR
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:997:27: error: 'RCC_APB1RSTR_CAN1RST' was not declared in this scope
         RCC->APB1RSTR |=  RCC_APB1RSTR_CAN1RST;
                           ^~~~~~~~~~~~~~~~~~~~
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:997:27: note: suggested alternative: 'RCC_APB1RSTR1_CAN1RST'
         RCC->APB1RSTR |=  RCC_APB1RSTR_CAN1RST;
                           ^~~~~~~~~~~~~~~~~~~~
                           RCC_APB1RSTR1_CAN1RST
3rdparty/libuavcan/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp:998:14: error: 'struct RCC_TypeDef' has no member named 'APB1RSTR'; did you mean 'AHB1RSTR'?
         RCC->APB1RSTR &= ~RCC_APB1RSTR_CAN1RST;
              ^~~~~~~~
              AHB1RSTR
make: *** [build/obj/uc_stm32_clock.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [build/obj/uc_stm32_can.o] Error 1

Regards,

JeremySavonet, 2019-03-21:
This could be a patch if needed:

```
diff --git a/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp b/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp
index 2fe8ad67..ebd21a69 100644
--- a/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp
+++ b/libuavcan_drivers/stm32/driver/src/uc_stm32_can.cpp
@@ -993,9 +993,9 @@ void CanDriver::initOnce()
         modifyreg32(STM32_RCC_APB1RSTR, RCC_APB1RSTR_CAN2RST, 0);
 # endif
 #else
-        RCC->APB1ENR  |=  RCC_APB1ENR_CAN1EN;
-        RCC->APB1RSTR |=  RCC_APB1RSTR_CAN1RST;
-        RCC->APB1RSTR &= ~RCC_APB1RSTR_CAN1RST;
+        RCC->APB1ENR1  |=  RCC_APB1ENR1_CAN1EN;
+        RCC->APB1RSTR1 |=  RCC_APB1RSTR1_CAN1RST;
+        RCC->APB1RSTR1 &= ~RCC_APB1RSTR1_CAN1RST;
 # if UAVCAN_STM32_NUM_IFACES > 1
         RCC->APB1ENR  |=  RCC_APB1ENR_CAN2EN;
         RCC->APB1RSTR |=  RCC_APB1RSTR_CAN2RST;
diff --git a/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp b/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp
index 1a4cd1e6..120343e6 100644
--- a/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp
+++ b/libuavcan_drivers/stm32/driver/src/uc_stm32_clock.cpp
@@ -38,10 +38,10 @@
 # endif

 # if UAVCAN_STM32_TIMER_NUMBER >= 2 && UAVCAN_STM32_TIMER_NUMBER <= 7
-#  define TIMX_RCC_ENR           RCC->APB1ENR
-#  define TIMX_RCC_RSTR          RCC->APB1RSTR
-#  define TIMX_RCC_ENR_MASK      UAVCAN_STM32_GLUE3(RCC_APB1ENR_TIM,  UAVCAN_STM32_TIMER_NUMBER, EN)
-#  define TIMX_RCC_RSTR_MASK     UAVCAN_STM32_GLUE3(RCC_APB1RSTR_TIM, UAVCAN_STM32_TIMER_NUMBER, RST)
+#  define TIMX_RCC_ENR           RCC->APB1ENR1
+#  define TIMX_RCC_RSTR          RCC->APB1RSTR1
+#  define TIMX_RCC_ENR_MASK      UAVCAN_STM32_GLUE3(RCC_APB1ENR1_TIM,  UAVCAN_STM32_TIMER_NUMBER, EN)
+#  define TIMX_RCC_RSTR_MASK     UAVCAN_STM32_GLUE3(RCC_APB1RSTR1_TIM, UAVCAN_STM32_TIMER_NUMBER, RST)
 # else
 #  error "This UAVCAN_STM32_TIMER_NUMBER is not supported yet"
 # endif
```

pavel-kirienko, 2019-03-24:
Thanks, @JeremySavonet! The patch would break other MCUs though. Would it not be possible to implement the fix in a way that wouldn't have any destructive consequences for other MCUs? We would gladly accept that into the upstream.

JeremySavonet, 2019-03-25:
Here is a cleaner fix without breaking build for other MCUs:
https://github.com/UAVCAN/libuavcan_stm32/pull/3

pavel-kirienko, 2020-03-07:
Fixed in #3

closed 2020-03-07 by pavel-kirienko