OpenCyphal-Garage/platform_specific_components #9: Create STM32H743 Driver

thirtytwobits, 2020-04-16:
Create a driver for FD reference board: STM32H743 Nucleo-144 (ST micro)

dagar, 2020-09-30:
For reference @JacobCrabill added this legacy libuavcan stm32h7 driver in PX4. https://github.com/PX4/Firmware/tree/master/src/drivers/uavcan/uavcan_drivers/stm32h7/driver

pavel-kirienko, 2020-09-30:
Nice.

@dagar can we adapt it to support CAN FD (it seems to support only Classic CAN at the moment) and refactor it to remove dependencies on Libuavcan v0? Much like we did for bxCAN. That would be super useful.

JacobCrabill, 2020-09-30:
I'm actually (slowly) going through the process of adding Socket can support for the H7 series in Nuttx; I can make my WIP public if it speeds things up.

pavel-kirienko, 2020-09-30:
It might, but what we really need is someone willing to do the actual work of extracting the low-level logic from the v0/NuttX driver into a separate reusable component.

JacobCrabill, 2020-10-01:
@pavel-kirienko I guess your goal here is to have a standalone set of files to put into any generic project, not just NuttX / PX4?  I think the long-term goal for PX4 is to simply have all the drivers built into NuttX; that's the approach taken by @PetervdPerk-NXP for the UAVCAN v1 test setup on the NXP CAN boards, and it makes sense to me.  I don't think it would be too difficult to take what I put together (based on the outline of your bxCAN driver) and make it libuavcan-independent.

agribov, 2023-11-06:
Has anyone developed a fdcan driver compatible with libuavcan/libcanard?

PetervdPerk-NXP, 2023-11-06:
Under NuttX there's Cyphal support which uses the SocketCAN api.
S32K1XX, S32K3XX and IMXRT suport CAN FD through this.
https://github.com/apache/nuttx-apps/blob/master/examples/opencyphal/canard_main.c

aentinger, 2023-11-06:
>Has anyone developed a fdcan driver compatible with libuavcan/libcanard?

Are you looking for a **CAN FD HAL driver**, i.e. for specific MCU platform?
Or do you want to use libcanard with CAN FD?

The 107-Arduino-Systems would support [CAN FD](https://github.com/107-systems/107-Arduino-Cyphal/blob/main/src/Node.hpp#L72) frames via libcanard (and can be cross-compiled via `CMake` for any target platform - not limited to the Arduino ecosystem).