OpenCyphal-Garage/platform_specific_components #11: STM32 example missing for V1

TomDeRybel, 2020-08-25:
Hello Pavel,

I'm trying to implement Libcanard on an STM32 platform. Unfortunately, the examples for anything but socketcan seem to be missing from master. I found the legacy V0 implementation, but it does not seem to be compatible with the current V1 code.

Do you have a compatible version available?

Best regards,
Tom

pavel-kirienko, 2020-08-25:
The driver you are looking for does not exist yet; however, we have its exhaustive specification in the form of a well-documented header file: https://forum.uavcan.org/t/libcanard-v1-stm32-driver-design-guidelines/798.

We need somebody to write the implementation for the header file. Can you please help us with that?

TomDeRybel, 2020-08-26:
Hello Pavel,

Well, that is rather a tall order for my limited programming skills. However, I'm giving it a try and see where we end-up. 

I've started porting the old driver to conform to the new header today. I got a few functions working already, in fact. Though, I will only be able to test CAN1. I'm writing it with CAN2 support included, but, well, it compiles...

One thing I've already bumped into is the need for a microsecond accuracy timer for usleep() and message time-stamping. The header states this should never overflow. This means a 64-bit, monotonic counter for all intends and purposes.

The trouble is that such a thing doesn't exist on the STM32. The best thing standard available is the SysTick, which is 32 bit and generally set to 1 kHz. Good enough for the usleep time-out (as it really only needs loosely ms ticks), but not for the message time-stamping in my intended application (I prefer better than ms resolution on that).

So, what I'm working on now is this for a uint64_t microsecond timer:
- 16 bit HW timer with overflow interrupt, running at 1 MHz. -> 16 LSB
- This interrupt increments a 64-bit SW timer at ~15 Hz. -> 48 MSB

A couple functions read these and combines it into an uint64_t for time-stamping and sleep.
And, of course, this thing is a mess of potential race conditions....

Any thought s on how to properly handle this?

pavel-kirienko, 2020-08-26:
The driver does not need the timer and it certainly can't contain usleep or any other blocking activities. Your application may need that, but it's not part of the driver, is it.

The logic you are looking for is implemented in https://github.com/UAVCAN/platform_specific_components/blob/legacy-v0/stm32/libuavcan/driver/src/uc_stm32_clock.cpp and https://github.com/PX4/sapog/blob/master/firmware/src/motor/realtime/motor_timer.c; in both implementations, the race conditions are taken into account and proper avoidance measures are applied.

TomDeRybel, 2020-08-26:
Hm, that surprises me. I'm looking at the old driver, and L139 in canard_stm32.c clearly has usleep() in it. This is part of the initialization, waiting for the CAN to sync, so I suppose that blocking there can be tolerated. (I'm doing it by polling SysTick.)

Ah, ok. I've just noticed that I don't need to obtain the current time stamp inside of the bxCANPush() function. I missed that it takes that as an argument. Thus, yes, that means I won't need any time functions as part of the driver (outside of the init), as you say.

pavel-kirienko, 2020-08-27:
> I'm looking at the old driver, and L139 in canard_stm32.c clearly has usleep() in it. This is part of the initialization, waiting for the CAN to sync, so I suppose that blocking there can be tolerated.

Yes, indeed, that is the case. Above I spoke about the general runtime logic; initialization is a special case. A high precision is not required here, the objective is to break the initialization loop eventually to avoid stalling the driver if the controller or the network are malfunctioning. You can just use a busyloop (like in the Libuavcan STM32 baremetal driver), possibly using the CAN timestamp register for time measurement (not sure if it's feasible).

TomDeRybel, 2020-08-28:
Making progress...
- Completed the initial port and it compiles + runs (the parts tested thus far). Debugging starts on Monday.
- Just got the USB CAN interface going on the PC and I can now see what's going on. Also got the logic analyzer hooked to the bus.
- TX (from the STM32) seems to work fine, but it doesn't abort fully when there is nothing else on the bus. To be investigated.
- RX partly works. I think I've got a pointer address/value confusion somewhere for the payload data, and the extended can id is not masked properly.

This is as far as I've gotten this week.

Some questions:
- The old driver supported the standard can id and the extended can id. The new driver header seems to state that the extended can id is to be supported exclusively, and the standard can rejected. Am I reading this right?
- I've found the need to add the following defines to the bxcan driver, to be able to convert between the STM32 registers and a more standardized representation (as the old driver did it):
define CANARD_CAN_EXT_ID_MASK                      0x1FFFFFFFU
#define CANARD_CAN_STD_ID_MASK                      0x000007FFU
#define CANARD_CAN_FRAME_EFF                        (1UL << 31U)         ///< Extended frame format
#define CANARD_CAN_FRAME_RTR                        (1UL << 30U)         ///< Remote transmission (not used by UAVCAN)
#define CANARD_CAN_FRAME_ERR                        (1UL << 29U)         ///< Error frame (not used by UAVCAN)
I've changed the names to start with BXCAN_FRAME_, and added this to the C file. They are no longer part of the new libcanard, nor bxcan headers, but seem needed to get the data formatted right. Is this the right approach?

As to the busy wait: I've implemented it that way. Using the timestap register made me worry that the following from the MCU errata may be a problem (there has to be a time source somehow for the timestamp to work I presume?): 
---
bxCAN time-triggered communication mode not supported

Description: The time-triggered communication mode described in the reference manual is not supported. As a result,
timestamp values are not available. The TTCM of the CAN_MCR register must be kept cleared (time-triggered
communication mode disabled).

Workaround: None.
---

Have a nice weekend!

pavel-kirienko, 2020-08-28:
> The old driver supported the standard can id and the extended can id. The new driver header seems to state that the extended can id is to be supported exclusively, and the standard can rejected. Am I reading this right?

Yes.

> I've found the need to add the following defines to the bxcan driver, to be able to convert between the STM32 registers and a more standardized representation (as the old driver did it)

These definitions were needed because the old libcanard had the notion of standard frames and error frames, even though they were never used. The new libcanard does not require these definitions because it only recognizes extended frames. Also note that these definitions have nothing to do with bxCAN itself so even if you decided to keep them for whatever reason, you shouldn't put them into that header.

> As to the busy wait: I've implemented it that way. Using the timestap register made me worry that the following from the MCU errata may be a problem

Ok, this does look like a valid concern.

TomDeRybel, 2020-09-01:
Hello Pavel,

I'm making progress. It seems to send correct data (using the UAVCAN GUI Tool), receives it as well (up to a point, see below). I have not tested filters yet. But, I'm going to need a bit of help. I'm not really a programmer (nice way of saying "rank amateur"), so the following is probably trivial to you. 

I'm getting lost in my pointers, and it's preventing further testing and improving the driver.
In main.c L299 (pardon the mess in here): see the TODO. This is where it starts to go wrong.
In bxcan.c L577 (the other side of the thing): see the TODO. The other part where it goes wrong.
The files are attached.

All the other TODOs are mainly questions/remarks. Any feed-back or code changes there are definitely appreciated.

Otherwise, it's nearing a basic state of "done".

Best regards,
Tom

[bxcan_20200901.zip](https://github.com/UAVCAN/platform_specific_components/files/5156521/bxcan_20200901.zip)

pavel-kirienko, 2020-09-03:
Not bad.

One of the most important principles of API design is encapsulation: keep the implementation details of your module (driver in this case) away from the user's eyes. The user doesn't need to know how the assertion check macro is defined, or what is the value of `BXCAN_BUSYWAITMS`, because these are mere internal configuration details of your module. I doubt `BXCAN_CONFIG_DEBUG_INNER_PRIORITY_INVERSION` is needed there either -- whether the check is enabled should not affect the usage of the driver. Please move these things away from the header because **the header file defines the interface to your driver**.

For the same reason, please remove from the header the two extra inclusions you've added: `stddef.h` and `assert.h`. They should reside in the `.c` file because they do not participate in the definition of the module interface. I actually suggest you avoid the introduction of any changes to the original header file I shared on the forum unless there are any strong reasons for that, which we should discuss separately. A positive change you've introduced into the header is that you've replaced `\` with a `|` in a comment; the use of a backslash in a comment was a mistake on my part.

Please keep the code formatted and adherent to the Zubax coding conventions. I have noticed the following issues:

- The body of control flow constructs is not always a block statement. You should always use curly braces, it is mandated by MISRA and many other high-integrity coding standards.
- Do not use tabs.
- Use `//` comments instead of `/*` -- the original driver is non-compliant but it should be trivial to fix.
- Whenever you define a variable that is not intended to be changed, declare it `const`. This simple rule effectively protects you against a certain class of mistakes. For example, replace the following:

```diff
- static void processErrorStatus(volatile BxCANType* bxcan_base, volatile bool* error_can)
+ static void processErrorStatus(volatile BxCANType* const bxcan_base, volatile bool* const error_can)
```

Use Clang-Tidy to find these and similar issues automatically; you can use the configuration file from O1Heap or Libcanard, for example. Likewise, you can use Clang-Format to fix the formatting issues.

I see that you have taken a very diligent approach to the initialization timeout tracking. This is fine, but observe that you are introducing complexity into the driver configuration that is trivially avoidable: the parameter `BXCAN_BUSYWAITMS` can be redefined as the core clock rate, and the required arithmetic conversions (namely, the division by 2000) can then be done by the driver itself. Also, a more descriptive name is needed; an overly long name is better than an overly short one because generally, long descriptive names = clearer API.

I spot issues with error handling. A recurring pattern in your code is where you apply assertion checking to validate the input arguments. If the function is typed to return an error status, use it to report invalid arguments to protect against misuse in release builds where the assertion checks may be disabled. For example, instead of `BXCAN_ASSERT((iface_index >= 1) && (iface_index <= 2));`, check the value explicitly and return error if inadequate.

Also, there are orthogonality issues where you validate the value in one place and then make assumptions about it in a different place, which is error-prone:

```
if ((iface_index < 1) || (iface_index > 2))
{
    return false;
}
...
if (iface_index == 1)
{
    ...
}
if (iface_index == 2)
{
    ...
}
```

This can be restated as:

```
if (iface_index == 1)
{
    ...
}
else if (iface_index == 2)
{
    ...
}
else
{
    ... // flag error
}
```

Your code makes references to libcanard (including its v0 release). Please avoid this because the driver is supposed to be agnostic of the higher-layer logic.

There are some logical errors in the code although I did not dedicate much attention to those, I suppose you are still working on that; like this one:

```
if (extended_can_id >= convertFrameIDRegisterToCanard(bxcan_base->TxMailbox[i].TIR))
{
    return false;    // Error -- you can't return here because you would skip the error handling logic after the loop.
}
```

[... 36 lines cut ...]

```c
uint8_t* out_ptr = (uint8_t*) out_payload;
*out_ptr++ = (uint8_t)(0xFFU & (rdlr >>  0U));
*out_ptr++ = (uint8_t)(0xFFU & (rdlr >>  8U));
*out_ptr++ = (uint8_t)(0xFFU & (rdlr >> 16U));
*out_ptr++ = (uint8_t)(0xFFU & (rdlr >> 24U));
*out_ptr++ = (uint8_t)(0xFFU & (rdhr >>  0U));
*out_ptr++ = (uint8_t)(0xFFU & (rdhr >>  8U));
*out_ptr++ = (uint8_t)(0xFFU & (rdhr >> 16U));
*out_ptr++ = (uint8_t)(0xFFU & (rdhr >> 24U));
```

In `main.c`:

> TODO: This goes horribly wrong in bcCANPop(). I'm lost in all this pointering around...

A bit too much indirection there. Use this:

```c
CanardFrame rx_frame = {0};            // zero initialization is recommended as a defensive tactic
// note that in the following the type is uint8_t[], not uint8_t*[].
uint8_t rx_frame_data[8] = {0};        // just a single {0} is enough to zero-fill everything
rx_frame.payload = &rx_frame_data[0];  // do not cast arrays to pointers
```

TomDeRybel, 2020-09-04:
Hello Pavel,

Thank you for the detailed and educational reply. I'll get back to working on the driver a week from now and use the intervening time to read-up on the things you've explained.

I'm getting an education here :-)

Best regards,
Tom

TomDeRybel, 2020-09-11:
Hello Pavel,

My schedule cleared-up and I was able to work on the code this week, see attached.

I have implemented all your comments and suggestions as well as I could. I'll work with the code over the next period as I build the rest of the application around it and will likely discover remaining issues. That said, It's probably a good time for an expert eye to go over it all in detail.

There are two questions left in the bxcan.c file, both minor.

Regarding your detailed response, I'm taking the lessons to the other code I'm working on. There are a few points remaining:

> For the same reason, please remove from the header the two extra inclusions you've added: stddef.h and assert.h. They should reside in the .c file because they do not participate in the definition of the module interface.

stddef.h proved required on my platform for size_t, which is used in the header definition. The others are moved.

> Use Clang-Tidy to find these and similar issues automatically; you can use the configuration file from O1Heap or Libcanard, for example. Likewise, you can use Clang-Format to fix the formatting issues.

I'm having issues getting the Eclipse plug-in to work (using STM32CubeIDE right now). Running it manually will take some doing to set-up. (Will give it a go at a later time. I got it to run, but it ended-up checking most of the ST HAL as well due to a required include, and that didn't go over so well...) That said, I read the Zubax C guidelines, and tried to conform as best as I could.

On the pointer issues, I've made the indicated changes, but there's still an issue. See: L304 main.c. I "fixed" it, but not in  a good way. That said, it transmits and receives properly now.

Have a nice weekend,
Tom

[bxCAN.zip](https://github.com/UAVCAN/platform_specific_components/files/5209424/bxCAN.zip)

pavel-kirienko, 2020-09-11:
> stddef.h proved required on my platform for size_t, which is used in the header definition. The others are moved.

This is correct. I missed this dependency, my bad.

> I'm having issues getting the Eclipse plug-in to work (using STM32CubeIDE right now). Running it manually will take some doing to set-up. 

Getting it to work with the IDE might be a waste of time. This is a very simple tool that is trivial to run from the command line, it's probably best to stick to that seeing as the use case at hand is so unsophisticated. Just put the config files into the same directory with the sources, or somewhere above in the directory tree, and run:

```sh
clang-format -i -fallback-style=none -style=file --verbose bxcan.[ch]
clang-tidy bxcan.c
```

Please expand all tabs into spaces (there are currently tabs in `bxcan.c`).

---

I don't think the following is wise because it introduces an extra dependency that is easily avoidable. I think your original approach with a macro was better (sans the minor issues I mentioned with the name and the derivation logic).

```c
#include "system_stm32l4xx.h"   // For CMSIS SystemCoreClock. Will depend on your platform.
```

---

```c
// TODO: I would suggest to add the comment just below in this file, or the header. As a
// newcomer to the project, I had no idea how the extended_can_id value was supposed to be
// formatted until I took the code apart and referenced it back to the bxCAN registers.
// This is also where the many references to Libcanard came from originally: I was trying
// to link this to written documentation. My €-cent only later dropped that the format was
// that of the CAN 2.0B standard (logically, in retrospect).

/// This bxCAN driver exclusively supports extended-id frame format (29-bit) data frames.
/// There is no support for the classic-id (11-bit) data frames, remote frames, overload
/// frames, or error frames. The 29 bits of the extended identifier values used in this
/// driver match the order of the CAN protocol, as described in the CAN 2.0B standard.
```

Good, let's do this. I think it belongs somewhere in the header because that's where the docs are.

---

Here's a minor nitpick that is not really important but why not mention it. Occasionally, automatic formatters break neatly formatted fragments because they don't understand the semantics of the code:

```c
    if ((timings.bit_rate_prescaler < 1) || (timings.bit_rate_prescaler > 1024) ||
        (timings.max_resync_jump_width < 1) || (timings.max_resync_jump_width > 4) || (timings.bit_segment_1 < 1) ||
        (timings.bit_segment_1 > 16) || (timings.bit_segment_2 < 1) || (timings.bit_segment_2 > 8))
    {
        return false;
    }
```

You can force the auto-formatter to respect your preferred line breaks by sprinkling empty comments at the end of the line:

```c
    if ((timings.bit_rate_prescaler < 1) || (timings.bit_rate_prescaler > 1024) ||     //
        (timings.max_resync_jump_width < 1) || (timings.max_resync_jump_width > 4) ||  //
        (timings.bit_segment_1 < 1) || (timings.bit_segment_1 > 16) ||                 //
        (timings.bit_segment_2 < 1) || (timings.bit_segment_2 > 8))
    {
        return false;
    }
```

---

The pointer type of the second argument to `void processErrorStatus(volatile BxCANType* const bxcan_base, volatile bool* const error_can)` does not need to be volatile because its state is fully managed by your logic (as opposed to `volatile BxCANType` which can be updated by external forces so it has to be volatile).

---

```
            // TODO: In case an STD, ERR or RTR frame was received, one option is to return
            // success (We read the data incoming data just fine), but set *error_can to
            // indicate that the result is not good as we don't handle those frames properly.
            // I went with another option: silently ignoring these frames by returning false.
            // Any preference?
```

I don't see any issues with just dropping these frames. There may be other traffic on the bus that is of no interest to the application. While normally such traffic should be discarded by the acceptance filters, I see no reason to require that, so your current policy is sensible, except for one issue: if I understand the logic correctly, when you ignore a frame, you are supposed to continue to the next iteration instead of returning early, because that may cause you to skip a correct frame waiting in the next FIFO.

---

There is an elephant :elephant: in the room that I missed in the first review: indexes in C start with zero, so when we say "interface index", we assume that CAN1 has index 0, and CAN2 has index 1. The way you implemented it right now seems counter-intuitive.

---

```c
    // TODO: the pointers still go wrong. The commented-out line fails, the one below works.
    // In the line commented-out, the address in &rx_frame.payload gets changed with the data.
    // In the active line, the data in goes where it has to go. So, something with the pointers
    // not going deep enough into the struct (don't know how to express it technically)....
    // I want this to work with the CanardFrame struct (as in the commented-out line).
    //if (bxCANPop(1, &rx_frame.extended_can_id, &rx_frame.payload_size, &rx_frame.payload))
    if (bxCANPop(1, &rx_frame.extended_can_id, &rx_frame.payload_size, &rx_frame_data[0]))
```

The problem in the commented line is that `rx_frame.payload` is already a pointer. When you take a pointer to a pointer, you get one more level of indirection than necessary, and you end up with your payload written into the struct where there should be a pointer (not the value it points to). The correct form is like you wrote but without the indirection:

```c
bxCANPop(1, &rx_frame.extended_can_id, &rx_frame.payload_size, rx_frame.payload)
```

TomDeRybel, 2020-09-15:
Hello Pavel,

I've made the changes and gotten the formatting tools to work. The spurious tabs should be gone as well.

With respect to the clang-tidy messages, I've ignored all the ones complaining about magic numbers (given the nature of this C code), as well as one regarding the use of the assembly NOP instruction in the busy wait (though, if we really want to nail this down, it could become perhaps a pointless mathematical operation of sorts, say a decrement inside of a while loop, and hoping that the compiler isn't smart enough to see that). All the other messages were modified to satisfy the tool (and it caught a few mistakes on my part as well).

The :elephant: had also caught my attention, but being a hardware guy, I had consistency with the ST datasheets in mind and didn't think more of it. In any case, a :mouse: was applied, as well as a small comment added to the header to point-out the numbering mis-match between bxCAN and the ST documentation for the macrocells. If anything, it has made the validation logic a bit lighter, saving some cycles.

The filter code is now also tested. There is one issue in the documentation: L97 bxcan.h - setting the filters to 0,0 means accept all, not reject all (confirmed by tests). A proposal could be to use 0,0x1FFFFFFF. Though, messages with an extended id of 0 would still pass (if such messages are valid/can exist?). 

At this point, I have two of my devices talking on the bus, sending and receiving heartbeat messages. I've also got UAVCAN GUI Tool looking at the traffic, and generating V0 NodeStatus messages to cause some contrarian traffic on the bus. Everything seems to be working fine.

There is one TODO question that came-up working on the filters. See bxcan.c, L303. I would be inclined to do a check of the interface number and just return if out of range. Further tests of if that interface is valid might be skipped then?

There is another potential trap in the whole interface numbering detection: if CAN2 is selected on hardware that only has CAN1, that will not be caught and will result in "interesting" behaviour. Not sure to what degree of validation I have to go. If this is an issue, one option could be to have a define set to the number of interfaces, and checking. Of course, that also fails if it is not set
right. Perhaps the hardware can be polled. But, all of that starts to feel a bit excessive. Your thoughts?

One last thing: those pointers... The changes you explained bring me back to pretty much where I started. But, the compiler doesn't like it... (Given that I just got a call-back function pointer registration table to work right apparently proves little. I'm one step away from breaking-out my crayons and drawing the pointers and addresses in stick figures.) Though, what's left is more of a typing issue, it seems?

> CanardFrame rx_frame         = {0};  // Zero initialization is recommended as a defensive tactic.
> uint8_t     rx_frame_data[8] = {0};  // Standard CAN MTU = 8 bytes.
> rx_frame.payload             = &rx_frame_data[0];`
> 
> // Try to receive any pending CAN messages.
> // TODO: Still one issue:
> // warning: passing argument 4 of 'bxCANPop' discards 'const' qualifier from pointer target type
> // note: expected 'void * const' but argument is of type 'const void *'
> // if (bxCANPop(0, &rx_frame.extended_can_id, &rx_frame.payload_size, rx_frame.payload))
>  if (bxCANPop(0, &rx_frame.extended_can_id, &rx_frame.payload_size, &rx_frame_data[0]))

Other then that, various steps closer to completion.

[bxcan.zip](https://github.com/UAVCAN/platform_specific_components/files/5225091/bxcan.zip)

pavel-kirienko, 2020-09-15:
> With respect to the clang-tidy messages, I've ignored all the ones complaining about magic numbers (given the nature of this C code), as well as one regarding the use of the assembly NOP instruction [...]

You can tell the tool to ignore certain classes of errors by disabling the appropriate checks in the configuration file. The tool emits the name of the applicable options in the error output. For example, in the error message below it says that the diagnostic can be suppressed by adding `-llvm-qualified-auto` (yes, with the leading `-`) to the config file:

```
/home/travis/build/UAVCAN/libcanard/tests/test_public_rx.cpp:80:10: error: 'auto _session' can be declared as 'auto *_session' [llvm-qualified-auto,-warnings-as-errors]
    for (auto _session : ins.getInstance()._rx_subscriptions[1]->_sessions)
         ^
```

Regarding the NOP, it's not hugely important because by design the driver is intended to run on a very specific platform so it's not portable, but it is recommended nevertheless to avoid it. It can be safely dropped if you make the counter variable volatile -- this way the compiler will not optimize it away. Up to you.

> The elephant had also caught my attention, but being a hardware guy, I had consistency with the ST datasheets in mind and didn't think more of it. In any case, a mouse was applied, as well as a [...]

Regarding the issue at hand --- great.

Regarding your last remark --- it's not to patronize but are you sure you are setting your priorities correctly? Computational throughput and clarity of expression are often found on the opposite ends of the spectrum. Some industries tend to value one, others gravitate towards the opposite, depending on the trade-off between the cost of the execution time and the cost of unspecified behaviors. Complex videogames and airliner FBW systems can be thought of as polar opposites in this regard.

[image]

In our field, predictability and functional correctness are paramount, so one's first concern should be whether the thing you wrote is easy to understand rather than how many cycles it takes to execute it. The former is because complex things overload our limited cognitive abilities, increasing the chances that a mistake might creep in unnoticed.

> The filter code is now also tested. There is one issue in the documentation: L97 bxcan.h - setting the filters to 0,0 means accept all, not reject all (confirmed by tests). A proposal could be to [...]

The idea behind `{0, 0}` was that it is to be implemented by the driver by simply disabling the filter whose configuration is set to that. This is because the application needs a convenient way of telling the driver that it needs fewer filters than there are available in the hardware (if such is the case). What you said is possible but you have correctly noticed the undesirable corner case and also it introduces an extra magic number, so I think the approach it is described in the API docs is superior.

> There is one TODO question that came-up working on the filters. See bxcan.c, L303. I would be inclined to do a check of the interface number and just return if out of range. Further tests of if that interface is valid might be skipped then?

Okay but you can also do what you described without violating the single-return requirement of MISRA by using the iface index to compute the filter index offset early on and then simply add that to your `i`. At any rate, whatever you prefer is OK in this case.

> There is another potential trap in the whole interface numbering detection: if CAN2 is selected on hardware that only has CAN1, that will not be caught and will result in "interesting" behaviour. [...]

In the absence of the means of querying the hardware about the number of available interfaces, we will have to follow the policy known as "garbage in -- garbage out". Maybe put a cautionary note somewhere in the docs.

> warning: passing argument 4 of 'bxCANPop' discards 'const' qualifier from pointer target type

Ah yes, indeed, I haven't noticed. The type system of C is too limited to express certain things concisely; what you see here is an artifact of that limitation. You have two options:

- Remove the const qualifier explicitly: `(void*) rx_frame.payload`
- Use the pointer before the const qualifier is applied: `&rx_frame_data[0]`

What C programmers typically do in such cases is that they don't bother with immutability, usually making pointers non-const. This avoids such undesirable corner cases. I think that in this specific case though it's justifiable because due to the semantics of this API, the non-const pointer to the storage is always available to the user. In a langauge with a decent (Rust, D, C++) type system it would be possible to express the immutability without compromising on edge cases.

pavel-kirienko, 2020-09-15:
When the latest changes are finished, I think it's probably time for you to open a pull request with your changes, would you agree?

When doing that, please make sure to add your clang-tidy and clang-format configs next to the sources of your driver. We'll make them run in CI later.

TomDeRybel, 2020-09-16:
Hello Pavel,

Yes, I think the point has come where further working on it by me is going to make it worse. 

I have prepared the clang files. Only the clang-tidy needed some changes. I will open the pull request on Friday.

Until then, I have finally understood what you mean with MISRA single-return and re-worked everything (except the timing calculation) to comply as well as I could.

The busy-wait was re-worked again. The main reason is that other compilers may not use the same decorators, and I would like this thing to be portable in that sense. I also timed the loop (logic analyser) and checked the assembly. Turns-out it's 7 clock cycles (though, in debug). This silly thing is harder than it should be. (It would be nice if there was a truly standard, non-configurable, 1µs/tick HW timer register to read on the chip, at a standard, vendor-agnostic address...)

The filter code is changed to accept {0, 0} as "reject". I had not understood what you had in mind in the header. It does, indeed make for a cleaner API.

A define + checks to set the number of available HW interfaces was added. Mainly because I've shot myself in the foot this way already.

Dog and Pointer show: Thanks again for clarifying. I've gone with the cast to (void*), mainly for symmetry with the rest of the function call. In any case, both options work and I see clearer now (though not quite 100% yet).

On high-reliability: Re-reading my remark, I see that i really missed the ball. I wanted to say that, as a bonus, it saves a few cycles. Not as the prime objective. (I'm adding the image to my teaching collection. Good to make a point in the instrumentation and high-voltage course. My favourite is Weinberg’s Law: "If builders built buildings the way programmers wrote programs, then the first woodpecker that came along would destroy civilization.")

The main reason I choose UAVCAN as the core communication (and more) technology, and essentially bet my fledgling company on it, is precisely the focus on high-rel and well-researched foundations. I do not intend to undermine all of that by messing-up the driver. In fact, one of the reasons I said yes to writing it was to learn more about this way of working and in the process get better. Though, it is hard to know what I don't know, so I'll also admit to not always seeing the "why" behind your requests and explanations, or understanding what you intend and, as such, going my merry own way on the road to perdition.

Anyhow, attached is the current state of affairs. Friday, I'll try that pull request (first time).

Best regards,
Tom

[bxcan.zip](https://github.com/UAVCAN/platform_specific_components/files/5233210/bxcan.zip)

pavel-kirienko, 2020-09-29:
Fixed in #12

closed 2020-09-29 by pavel-kirienko