OpenCyphal-Garage/platform_specific_components #13: Continuous integration

pavel-kirienko, 2020-09-19:
We should set up CI that would run test suites on hardware or at least in a simulated environment. Any help to that end would be welcome.

pavel-kirienko, 2020-09-29:
> Regarding #13, I've given it some thought over the weekend. Where I may be willing to contribute is in (custom) test hardware for the project (Such as AVR, STM32, etc... (FD)CAN-enabled boards with the required diagnostics connections, a bus disturbance generation device, etc...) However, explicitly not in writing code. In any case, I'm willing to discuss a role in this.

_Originally posted by @TomDeRybel in https://github.com/UAVCAN/platform_specific_components/pull/12#issuecomment-700713737_

Test boards should not be a problem because existing dev boards seem to fit the purpose well. Controlled disturbance generation, however, is highly relevant. Can you please elaborate what do you have in mind, specifically?

At any rate, though, we will not be able to move this forward until we have someone willing to contribute to the software configuration part (set up the CI pipelines, the test machine, etc).

TomDeRybel, 2020-10-01:
What I was thinking of is this: with many CAN bus controllers, it is not possible to do overly stupid things on the bus. Yet, things like a stuck wire, bad connection, nodes with a hardware failure that generate nonsense, and EMC issues can impair bus traffic. While the CAN bus controller will reject such packets (CRC), they are still gone.

Having a device that can generate such bus issues would offer an opportunity to test error handling and recovery at low and high levels.

The device, in a simple form, could consist of a P and N mosfet on each CAN bus line, connected to VCC and GND, respectively, through a small resistance. The resistor is chosen to be able to over-power the CAN bus, but not so low that things get damaged. This allows for both high and low bus levels (though, only low or high would be enough to disrupt the bus).

Each of the four mosfets is connected to an individual I/O pin of a microcontroller. This allows for a tri-state condition (all mosfets off), device fully off-bus. The bit patterns are pre-computed and, using DMA from memory, applied to the I/O pins. 

Simultaneously, the bus levels are also monitored (using a CAN transceiver), so simple triggering can be applied (eg: falling-edge after a certain time-out) if this is relevant.

The simplest (just interfering blindly with the bus) device could be using a USB/Serial dongle, and using the handshaking pins as control lines, plus two N-mos+resistor to GND. That would interrupt all bus traffic, and involves no firmware at all, and can be driven trivially from the host. Slightly fancier is using one of the more recent FTDI chips in GPIO mode (there is Python user-land driver that works well on Linux and not too badly on Windows).

closed 2023-01-28 by pavel-kirienko