OpenCyphal-Garage/platform_specific_components #14: [UAVCAN/CAN] Driver for TI TCAN4550

pavel-kirienko, 2020-09-21:
Many popular MCUs lack built-in CAN FD-capable controllers, which limits the uptake of CAN FD in new products. There is an interesting IC that could help new hardware revisions or new designs adopt CAN FD without having to port an existing codebase to a new platform: [TI TCAN4550](https://www.ti.com/product/TCAN4550) -- a single-chip CAN FD controller with a built-in 5 Mbps transceiver interfaced with the MCU via SPI:

[image]

It is also appealing price-wise, being available from DigiKey for $1.5 @ 1k pcs, which is comparable to a regular standalone CAN transceiver.

This is a very promising chip and it would be nice to have portable and reusable drivers for it here.

Linjieqiang, 2020-10-11:
What about the MCP2518FD chip?

pavel-kirienko, 2020-10-11:
As far as I know, for MCP2518FD there are no AEC-Q/ISO26262 qualified
options, which is likely to be a deal breaker for some high-integrity
systems.

TCAN4550-Q1 is functionally identical to TCAN4550 plus safety
qualifications.

AGlass0fMilk, 2021-06-05:
Just found this while searching for in stock TCAN4550 parts...

I have modified the TI driver to support multiple TCAN instances and there's an implementation for Mbed-OS you can reference here:

https://github.com/AGlass0fMilk/mbed-tcan455x

I also designed [an open-source breakout board](https://www.theshackonwheels.com/getting-started-with-the-tcan455x-bob.html) for this part but thanks to COVID it won't be in stock again until 2022 🙃

Eventually I plan to make a group buy to produce a decent quantity of the breakout boards.

Just finding this project and I think it will work nicely for my CAN-based devices :)

Let me know if there's any interest.