OpenCyphal-Garage/platform_specific_components #15: [UAVCAN/CAN] for Arduino

aentinger, 2020-10-01:
Following up on my comment [here](https://forum.uavcan.org/t/libcanard-v1-stm32-driver-design-guidelines/798/3) I'm wondering about what's the best way to achieve this? There is currently no top-level `<CAN.h>` in the Arduino environment as there is for UART (`<Serial.h>`), I2C (`<Wire.h>`) and SPI (`<SPI.h>`). However, I could rather quickly donate a working Arduino Library intended for interacting with the (old but popular) `MCP2515` CAN communication chip. But how should the folder structure be set up?
```bash
arduino
|-- libcanard
   |-- arduino (for eventual top-level CAN support?)
   |-- mcp2515
```
What do you think @pavel-kirienko ?

pavel-kirienko, 2020-10-01:
I am beginning to think that the `libcanard` subdirectory might a bit of a misnomer in some cases because, for example, the bxCAN driver contributed by TomDeRybel is completely decoupled from Libcanard -- it is very generic and can be used in a completely standalone application (or even with libuavcan, perhaps, although it would require some serious bolting-on). Maybe in such cases, we should say `generic` instead of `libcanard` or something else? Or would that introduce confusion?

But anyway, following the currently practiced approach, I think just `arduino/libcanard/mcp2515` should be fine.

aentinger, 2020-10-01:
Are submodules in order too? Then I could just relicense [107-Arduino-MCP2515](https://github.com/107-systems/107-Arduino-MCP2515) to MIT and link it in. The library was written with being used with `libcanard` in mind, I could add an example demonstrating how to use the library in the presence of the `libcanard` defined types. What do you think? (A submodule has the added benefit that I need to fix issues only in one place).

pavel-kirienko, 2020-10-01:
That question made me realize something.

The code we keep here is not an actual product by itself, it is rather a pile of useful things that one can ruffle through in search of anything reusable. Meaning that this repository is never supposed to become part of anything wholesale. Embedding a library as a submodule amounts to just using this repository as a collection of references to external components intended to serve the same purpose. Continuing this line of thought I notice that it is just like adding a section "See Also" in the README pointing to various parts and components of the same nature.

I mean, adding a submodule makes sense if it is actually used for one of the drivers or test suites that are stored here, such as the case with libcanard that is submoduled to be used for building the unit tests for the SocketCAN driver. But if the submodule is a driver itself, then why have it, should we not just list the original repo in the docs?

aentinger, 2020-10-01:
I agree with you that this repository really starts to resemble a collection of useful code pieces and it might therefore be useful to just link towards the original repositories. As I see it, there's one small advantage of including submodules as opposed to linking to the original driver repository: If the original repository is deleted or turned private your link will point _nowhere_. Having the foreing repository included as submodule guarantees that you'll always have at least one copy at hand.

pavel-kirienko, 2020-10-01:
>Having the foreing repository included as submodule guarantees that you'll always have at least one copy at hand.

That's not how submodules work. A submodule is just a link maintained and used by git, so when the origin is gone, it's dead. :skull_and_crossbones: You probably are talking about git-subtree.

So which one do you prefer: [git-subtree](https://github.com/UAVCAN/pydsdl/blob/9a96baf67dc5957b7ab86762cc80e308a4c6be29/update_third_party_subtrees.sh#L15-L16) or a new section in the README? Or something else maybe.

aentinger, 2020-10-01:
Tbh I didn't know that about submodules. [git-subtree](https://www.atlassian.com/git/tutorials/git-subtree) looks like the way to go to me, although I've not used them myself yet.

pavel-kirienko, 2020-10-01:
Subtree it is then.

aentinger, 2020-10-01:
Hi @pavel-kirienko :wave: I've completed everything - You can git-tree-away using https://github.com/107-systems/107-Arduino-MCP2515 master:head.

pavel-kirienko, 2020-10-01:
I thought your preference would be to use subtree. If you are inclined to leave it to me, I think it's more sensible to add a See Also section in the README. Even if the link rots away one day, it will still be possible to find the project by its name online.

aentinger, 2020-10-01:
I think that git-tree is the sensible solution from a future-proof POV, however I would have left the git-tree implementation to you. Really, we can do what-ever floats your boat. I'm between submodules because it's kind of elegant and git-trees because it keeps one copy safe. The one thing I don't really like is the link approach. If you look into a repository you want to see code imho.

aentinger, 2020-10-06:
Just seeing that I mistyped above, I meant `git subtree` of course. :facepalm:

PR #16 (2020-10-12): Externally maintained third-party platform-specific components
This is a follow-up to https://github.com/UAVCAN/platform_specific_components/issues/15#issuecomment-702126881.

We need a third opinion so I would like to summon @thirtytwobits and let him choose how do we handle the inclusion of externally maintained third-party drivers:

- link from README like I am proposing here in this PR **[merge this PR if you choose this option]**
- use git-subtree **[reject this PR if you choose this option]**

Thanks.

Fixes #15
closed 2020-10-12 by pavel-kirienko