OpenCyphal-Garage/platform_specific_components #39: [UAVCAN/socketcan] Compilation error `CanardFrame`has no longer member `timestamp_usec`

aentinger, 2022-03-24:
This [line](https://github.com/UAVCAN/platform_specific_components/blob/master/socketcan/libcanard/src/socketcan.c#L262) causes a compile error when using with the [latest version of libcanard](https://github.com/UAVCAN/libcanard/blob/master/libcanard/canard.h#L176-L190).

pavel-kirienko, 2022-03-24:
See if this helps: #38

pavel-kirienko, 2022-03-24:
@aentinger would you like to volunteer some help with #38?

aentinger, 2022-03-25:
I might, but not prior to mid-june, as I'm currently pretty occupied with slapping together software for a certain event. It was during this slap-together process that this issue came to my notice but for now I'm fine just commenting the offending line out.

closed 2022-05-05 by pavel-kirienko