OpenCyphal-Garage/platform_specific_components #47: BxCAN1_Init() Function fix

Scoeerg, 2025-03-29:
Hi all,

I hope this helps some poor soul in the future. 

Working with STM32F466RE and [BX CAN](https://github.com/OpenCyphal-Garage/platform_specific_components/tree/master/stm32/libcanard/bxcan) instead of [HAL_CAN](https://stackoverflow.com/questions/56877821/how-does-the-can-bus-work-with-hal-in-stm32). 

`BxCAN1_Init()` did not initialize the CAN bus correctly. But here's the (yes, hard-coded) fix:

```c
void BxCAN1_Init(void){
    {
        // Enable CAN1 clock (same as HAL)
        __HAL_RCC_CAN1_CLK_ENABLE();

        __HAL_RCC_GPIOA_CLK_ENABLE(); // Enable GPIOA clock

        GPIO_InitTypeDef GPIO_InitStruct = {0};
        GPIO_InitStruct.Pin = GPIO_PIN_11 | GPIO_PIN_12; // PA11 = CAN1_RX, PA12 = CAN1_TX
        GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;          // Alternate function, push-pull
        GPIO_InitStruct.Pull = GPIO_NOPULL;              // No pull-up/down
        GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH; // Max speed
        GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;       // AF9 for CAN1 on STM32F446xx
        HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
    
        // Timing parameters from your HAL setup (500 kbps with PCLK1 = 42 MHz)
        BxCANTimings timings = {
            .bit_rate_prescaler = 6,     // Prescaler
            .bit_segment_1 = 11,         // TimeSeg1 (BS1)
            .bit_segment_2 = 2,          // TimeSeg2 (BS2)
            .max_resync_jump_width = 1   // SyncJumpWidth (SJW)
        };

        BXCAN1->MCR |= BXCAN_MCR_RESET;
        while (BXCAN1->MCR & BXCAN_MCR_RESET) {}
    
        // Configure CAN1 (iface_index 0) in normal mode
        if (!bxCANConfigure(0, timings, false)) {
            Error_Handler(); // Handle initialization failure
        }
    
        // Set filters to accept all frames (Cyphal will filter IDs)
        BxCANFilterParams filters[BXCAN_NUM_ACCEPTANCE_FILTERS] = {0};
        bxCANConfigureFilters(0, filters);
    }
}
```

which essentially mimics the behavior of HAL_CAN. It first initializes the pins:

```c
GPIO_InitTypeDef GPIO_InitStruct = {0};
    GPIO_InitStruct.Pin = GPIO_PIN_11 | GPIO_PIN_12; // PA11 = CAN1_RX, PA12 = CAN1_TX
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF9_CAN1;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
```

and later on resets the  bxCAN Master Control Register (BXCAN1) 

```c
BXCAN1->MCR |= BXCAN_MCR_RESET;
while (BXCAN1->MCR & BXCAN_MCR_RESET) {}
```

and waits for it to be actually reset. Since I am fairly new to this entire thing, I would love to hear your opinion @serges147 There's a good chance I just fuc*ed something up.

pavel-kirienko, 2025-03-29:
I am not sure where did you get `BxCAN1_Init` from, it is not part of this codebase.

A pull request adding a new item to the list of caveats here would be accepted, where the added item should mention that the CAN driver will not attempt to configure GPIOs (it's probably obvious anyway but won't hurt to spell it out):

https://github.com/OpenCyphal-Garage/platform_specific_components/blob/master/stm32/libcanard/bxcan/README.md#caveats

closed 2025-03-29 by pavel-kirienko

Scoeerg, 2025-03-29:
I got a little excited MY code works. I will leave it as a warning (or to copy-paste) to others. Initialize your darn GPIO pins correctly.